dewenni/ESP32-Jarolift-Controller #5: [BUG] Beim gleichzeitigen Ansteuern mehrerer Rollos aus HA wird nur der erste Befehl über Funk versendet

flushbug, 2024-12-18:
### Wurde der Fehler in der neuesten Version getestet?
- [x ] Ich nutze die v1.1.1.

### Fehlerbeschreibung
Ich habe vier Rolladen über die Home Assistant Gruppenfunktion zu einer Rolladengruppe verknüpft
Sobald ich einen Steuerbefehl absende wird dieser zwar von HA über MQTT versendet, der Jarolift Controller
Steuert jedoch nur den ersten Rolladen tatsächlich an. Siehe Log

```

[18.12.2024 - 20:04:42] I (APP-JARO): command STOP for channel 11 sent
[18.12.2024 - 20:04:42] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/9 | payload: STOP
[18.12.2024 - 20:04:42] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/10 | payload: STOP
[18.12.2024 - 20:04:42] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/11 | payload: STOP
[18.12.2024 - 20:04:42] D (APP-JARO): created devicekey low: 0x93a0fef1 // high: 0xba54365e
[18.12.2024 - 20:04:42] D (APP-JARO): get serial: 0x0000100b for channel: 11
[18.12.2024 - 20:04:42] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/12 | payload: STOP
[18.12.2024 - 20:04:36] I (APP-WEB): web-client disconnected
[18.12.2024 - 20:04:35] I (APP-JARO): Device Counter: 262
[18.12.2024 - 20:04:35] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/9 | payload: CLOSE
[18.12.2024 - 20:04:35] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/10 | payload: CLOSE
[18.12.2024 - 20:04:35] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/11 | payload: CLOSE
[18.12.2024 - 20:04:35] D (APP-JARO): created devicekey low: 0x93a0fef1 // high: 0xba54365e
[18.12.2024 - 20:04:35] D (APP-JARO): get serial: 0x0000100b for channel: 11
[18.12.2024 - 20:04:35] I (APP-MQTT): msg received | topic: jarolift/cmd/shutter/12 | payload: CLOSE
[18.12.2024 - 20:04:31] D (APP-WEB): Received - Element ID: p10_log_refresh_btn = true
``

dewenni, 2024-12-19:
Vermutlich ist das ein Timing-Problem. Durch die Gruppenfunktion werden die MQTT Nachrichten mehr oder weniger gleichzeitig gesendet. Der Sendeprozess über das CC1101 dauert dann aber etwas länger und dann gehen die anderen Kommandos wohl unter. Muss ich mir mal genauer ansehen und ggf. eine kleinen Kommando-Puffer einbauen.
Sollte lösbar sein.

wunderbaum, 2024-12-19:
Ja, habe ich mit der "alten" Jarolift auch schon. Ich habe einfach Pausen in die Automation eingebaut.

dewenni, 2024-12-21:
so, @wunderbaum @flushbug testet doch bitte mal das neue Release v1.2.0

Ich habe einen Kommando-Puffer eingebaut für die eingehenden mqtt Nachrichten und auch für die Abarbeitung der CC1101 Befehle. Die eignenden mqtt Nachrichten werden beim Empfang in eine Queue eingetragen und dann azyklisch abgearbeitet.
Bei der Abarbeitung werden dann die entsprechenden CC1101 Befehle in eine zweite Queue eingetragen. Das passiert aber immer noch sehr schnell und zu schnell für den CC1101.
Daher wird die CC1101 Queue nur alle 500ms abgefragt und enthaltene Befehle werden ausgeführt.
Damit sollte das Problem jetzt hoffentlich nicht mehr auftreten.

flushbug, 2024-12-21:
Funktioniert ausgezeichnet ! Klar, das Delay zwischen den Befehlen ist sichtbar aber macht in der Realität nichts aus.

closed 2024-12-21 by flushbug

dewenni, 2024-12-21:
Prima, danke für das schnelle Feedback.
Vielleicht kann man das auch noch etwas optimieren. Ich wollte jetzt erstmal die Funktion sicherstellen.
Ich glaube das Hauptproblem lag sogar an der MQTT Verarbeitung. Die aktuell statischen 500ms kann man ggf. noch optimieren. Aber wie du schon schreibst, eigentlich ist das ja egal wenn die etwas versetzt angesteuert werden.
Da wäre es schon interessanter wenn ich versuche die Gruppensteuerung zu implementieren die Jarolift ja auch grundsätzlich unterstützt. Das steht auch noch auf meiner ToDo-Liste.

flushbug, 2024-12-21:
Nachdem ich jetzt selbst die Software bauen kann (habe sogar gestern Nacht Amateurhaft versucht die queue selber einzubauen aber ohne zu testen) kann ich auch selbst versuchen mit dem Timing zu spielen/optimieren. Aber guter Punkt bzgl Gruppencommand, da muss es ja etwas geben denn meine Fernbedienung kann auch mehrere Rollladen gleichzeitig ansteuern und die bewegen sich exakt zur selben Zeit.

wunderbaum, 2024-12-21:
Bei mir wird es wohl leider erst nach den Feiertagen etwas. 
Will die Familie nicht mit Rollo Tests nerven 😉😂

dewenni, 2024-12-24:
Ich habe im aktuellen Release v1.3.0 auch die Gruppenfunktion implementiert.
Damit ist es nun wie bei den originalen Fernbedienungen möglich, Gruppen zu definieren und diese zeitgleich mit einem Befehl anzusteuern. Testet es gerne aus.