dewenni/ESP32-Jarolift-Controller #26: [BUG]

Blastmun, 2025-01-24:
### Is this bug present in the latest release?

v1.5.0

### Hardware Information

ESP32 D1 mini

### Bug Description

Hello, once again thank you for your project! I often encounter problems when I unplug it electrically and plug it back in, it goes back into setup mode with the default ssid and I have to connect to it and restart it in the GUI so that it recovers my configured wifi. I'm sorry I don't know much about coding but I notice in the config.h variables:
#define WIFI_RECONNECT 30000 // Delay between wifi reconnection tries
#define MQTT_RECONNECT 10000 // Delay between mqtt reconnection tries

What is this unit? Would it be possible to integrate the modification of its values ​​into the GUI? Maybe a change to 5/10 minutes would solve my problem automatically in case of problems.

### Additional Information

dewenni, 2025-01-25:
Hi,

I think there must be a different reason why the ESP in going into Setup-Mode.

Regarding your Questions:
When the ESP starts, it first tries to connect to the configured WiFi.
There are 5 active attempts to connect with a timeout of 30 seconds between each attempt.
After 5 attempts, the ESP will restart in "normal" Mode and try again. At this point there is no logic to switch to Setup-Mode.

Example:
```
I (APP-WDT): Watchdog timer initialized for Loop Task
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
E (44602) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
I (APP-SETUP): WiFi connection - attempt: 1/5
E (74601) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
I (APP-SETUP): WiFi connection - attempt: 2/5
E (104601) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
I (APP-SETUP): WiFi connection - attempt: 3/5
E (134601) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
I (APP-SETUP): WiFi connection - attempt: 4/5
E (164602) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
I (APP-SETUP): WiFi connection - attempt: 5/5
I (APP-SETUP): 
! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! !
I (APP-SETUP): Wifi connection not possible, esp rebooting...
I (APP-SETUP): 
! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! ! !
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
...
I (APP-WDT): enable cmd
I (APP-WDT): Watchdog timer initialized for Loop Task
I (APP-SETUP): WiFi Mode STA - Trying connect to: wrongSSID
E (44267) wifi:sta is connecting, return error
I (APP-SETUP): WiFi Mode STA - Trying connect to: asterix2
I (APP-SETUP): WiFi connection - attempt: 1/5
...
```

If the WiFi connection is successful, the same will happen for the mqtt connection.
5 attempts with a timeout of 10 seconds.

example:
```
I (APP-MQTT): MQTT setup done!
I (APP-MQTT): MQTT - connection attempt: 1/5
I (APP-MQTT): MQTT - connection attempt: 2/5
I (APP-MQTT): MQTT - connection attempt: 3/5
I (APP-MQTT): MQTT - connection attempt: 4/5
I (APP-MQTT): MQTT - connection attempt: 5/5
I (APP-MQTT): MQTT connection not possible, esp rebooting...
```

but also here is no logic to enter the Setup-Mode.

What could be a reason to enter the Setup-Mode?

1. config.json file could not be read
2. no SSID or WiFi Password is configured
3. WiFi and Ethernet is disabled in configuration
4. Double-Reset-Detection

The most likely solution is therefore probably ‘double reset detection’. This function also makes it possible to switch to setup mode manually. For example, if a WiFi configuration is stored but is incorrect. In this case, the ESP would keep trying to connect.
In this case, you can switch to setup mode by double-clicking the reset button.
How does this work technically?
The code checks whether less than 3 seconds have passed between the last restart and the current restart.

This can be achieved by triggering the restart twice within 3 seconds using the reset button. 

However, a ‘double reset’ could also be recognised if there is a problem with the power supply and the ESP therefore restarts twice.

Or if the ESP restarts twice within 3 seconds due to a software crash.

We therefore need to find out why the ESP switches to setup mode or why it recognises a double reset.

I'll see if I can enter the reason for switching to setup mode in the log so that you can see it in the WebUI logger after setup mode is active.

dewenni, 2025-01-26:
@Blastmun update mal auf das neue Release v1.5.1 und wenn der ESP das nächste mal in sen Setup-Mode wechselt, dann verbinde dich mal mit dem AccessPoint `SSID: ESP32-Jarolift / IP: http://192.168.4.1` und prüfe was im Log als Ursache für den Setup-Mode steht.

Klajan, 2025-02-01:
Ich habe ein ähnliches problem, die logs sagen es ist ein Double Reset.
Dieses Problem ist allerding auch schon bei einem restart aus der Web UI aufgetreten.

Version: v1.5.1

> [01.02.2025 - 00:11:37] I ESU-DRD: doubleResetDetected
> [01.02.2025 - 00:11:37] I MAIN: SETUP-MODE-REASON: DRD detected
> [01.02.2025 - 00:11:37] I CFG: LittleFS successfully started
> [01.02.2025 - 00:11:37] I MSG: LogLevel: ESP_LOG_DEBUG
> [01.02.2025 - 00:11:37] I SETUP: WiFi Mode: AccessPoint SSID: ESP32-Jarolift / IP: http://192.168.4.1
> [01.02.2025 - 01:11:37] I TELNET: Telnet Server running!
> [01.02.2025 - 01:12:47] D WEB: sending: /
> [01.02.2025 - 01:12:47] D WEB: sending: /main.css
> [01.02.2025 - 01:12:47] D WEB: sending: /main.js
> [01.02.2025 - 01:12:48] I WEB: web-client connected - IP:192.168.4.2
> [01.02.2025 - 01:12:48] D WEB: updateAllElements()
> [01.02.2025 - 01:12:51] D WEB: Received - Element ID: p10_log_refresh_btn = true

dewenni, 2025-02-08:
ok, ich werde mir dazu noch was überlegen um das robuster zu implementieren.
Eine einfache Möglichkeit wäre vermutlich, die Anzahl der notwendigen Neustarts in einem bestimmten Zeitfenster zu erhöhen.
Also das man z.B. den ESP 5x neu starten muss und das jeder Neustart innerhalb von 3 Sekunden erfolgen muss.
Erst dann wechselt der ESP in den Setup-Mode.

Generell halte ich es für wichtig, so eine Möglichkeit vorzusehen, damit man bei einer fehlerhaften Konfiguration (z.B. Tippfehler im WiFi Passwort) die Möglichkeit hat, den ESP wieder in einen Zustand zu bekommen, bei dem man per WiFi Access-Point zugreifen kann um die Einstellungen zu ändern.

Etwas praktikableres als das Auswerten von Neustarts habe ich dazu leider noch nicht gefunden.

dewenni, 2025-02-08:
ich habe es im Release [v1.6.0](https://github.com/dewenni/ESP32-Jarolift-Controller/releases/tag/v1.6.0) etwas überarbeitet.

Blastmun, 2025-02-27:
Entschuldigung, ich hatte Ihre Nachrichten nicht gesehen. Das Problem war nur während eines Neustarts vorhanden, da ESP32 wellig ist. Ich hatte keinen Neustart zu tun. Übergeben an 'Futter 1.5.0 bis 1.6.3 Das Update wird entsprechend und in jedem Fall hat das ESP32 korrekt neu gestartet

Klajan, 2025-02-28:
Auch ich kann bestätigen, dass es seit dem update keine Probleme mehr gibt, der ESP scheint trotztdem noch 2 restarts durchzuführen, aber bleibt nicht im setup modus hängen.

Es kann auch sein das mein chip problem hat mit der Spannung, laut der system info war der letzte restart Grund `Brownout (voltage dip)`

wunderbaum, 2025-02-28:
> Brownout (voltage dip)

Ein Kondensator über der Versorgungsspannung könnte für Abhilfe sorgen.

closed 2025-03-02 by dewenni