dewenni/ESP32-Jarolift-Controller #33: [BUG] Brown Out (voltage Dip), Web-Client disconnected, kein Senden/Empfang des CC1101 möglich

muede-git, 2025-02-16:
- Ich nutze die neueste Version 1.6.2.

### Hardware-Informationen

- ESP32-WROOM-32
- CC1101 von AINEWIOT (Bild vergleichbar zum Projektaufbau)

### Fehlerbeschreibung

Nachdem die Software kompiliert und hochgeladen wurde, startet der ESP32, die Einbindung in WLAN, MQTT und telnet funktioniert einwandfrei.
Beim Starten erscheint dann aber immer, unabhängig von welcher Qualität Netzteil ich dahinter hänge (bis zu 65W), ein brownout und danach die Meldung, dass der Web-Client nicht mehr verbunden ist. 
MSB und LSB sowie serial send sind aus meinem SDUINO aus fhem übertragen. Der CC1101 lauscht nicht. Es werden keine Signale empfangen, werden von der Originalfernbedienung, noch vom SDUINO NAO-Cul. Ein Anlernen ist somit auch nicht möglich.
Die Verbindung GPIO's wurde wiederholt geprüft und nun auch auf Lötpfosten gesetzt, dass keine kalten Lötstellen oder vergleichbar entstehen können.
Den Counter kann ich nicht an den aus fhem anpassen. Der Cache setzt den Wert beim Neustart immer wieder auf den Counter des ESP-Jarolift zurück.
Einen Screenshot des Debug-Logs habe ich beigefügt.

### Zusätzliche Informationen

[image]

dewenni, 2025-02-16:
Hi,

nach deinem Eintrag hier sind mir zwei Dinge aufgefallen die ich hoffentlich mit den gerade erstellen Release behoben habe.

**1) Device Counter lässt sich im WebUI nicht ändern.**  
Hier hatte ich intern von EEPROM auf NVS umgestellt und dabei aber noch einen kleinen Fehler eingebaut. Beim Ändern über das WebUI wurde vor dem speichern fälschlicherweise erst nochmal der Wert aus dem Flash geladen und damit die Eingabe wieder überschrieben. Sollte aber jetzt gelöst sein.

**2) Logging**  
Hier war es auch so, dass nach einer meiner letzten Änderungen scheinbar keine Log Meldungen mehr von der internen Jarolift-Controller Library durch kamen. Auch das sollte jetzt wieder passen. 

Bleibt noch das Thema mit dem Brownout (voltage-dip)  
Hier kann ich aber leider auch erstmal nur vermuten das es tatsächlich ein Problem oder Defekt gibt.
Normal ist das zumindest nicht. Bootet der ESP jetzt auch dann immer wieder neu, oder bleibt er dann auch stabil?

Teste mal das neue Release und gib nochmal Rückmeldung

muede-git, 2025-02-16:
Hi Sven, herzlichen Dank für deine schnelle Reaktion und Hilfe.
Der Counter kann geändert werden und zuhören tut er jetzt richtiger MSB, LSB und Serial jetzt auch.
Leider lernt er aber nicht an. Wo könnte ich meinen Fehler haben?
Der Brownout entsteht derzeit nur beim Start einmal; danach läuft er stabil.

dewenni, 2025-02-16:
ok, prima. 
Willst du nun deine alte Config übernehmen oder neu anlernen?

Ich würde erstmal testhalber mit einer neuen Seriennummer (z.B. 000010 ) und Device Counter = 0 beginnen.
Teste mal ob du damit einen Rollo anlernen kannst.

Wie genau lernst du an?
- Knopf am Motor drücken?
- Spannung vom Rollo AN/AUS?
- kopieren einer vorhandenen Fernbedienung?

muede-git, 2025-02-17:
Ich wollte von der bestehenden Fernbedienung anlernen, damit ich immer eine stabile Rückfallposition habe.
Damit wäre es die Übernahme der bestehenden Config der Originalfernbedienung und was fhem jetzt kann. Dem Grunde nach möchte ich die Jarolift dann nach homeassistant überführen.

dewenni, 2025-02-17:
Wie gesagt, ich würde vorschlagen erstmal zu testen ob dein Setup funktionsfähig ist.

- aktuelles Release verwenden
- dann im WebUI unter Logger auf "Debug" stellen
- vorhandene Fernbedienung in unmittelbare Nähe vom ESP bringen und ein Befehl auslösen (Hoch/Runter)
- prüfen ob im Log der Befehl erkannt wird (Es sollten 3 Meldungen erscheinen)

dann mal mit einer neuen Seriennummer (Empfehlung 000010) und Device Counter = 0 versuchen einen Rollo anzulernen.

muede-git, 2025-02-21:
Super läuft mit der 000010. Danke Dir

muede-git, 2025-02-22:
Ich nutze aus der fhem Diskussion um die sduino-Integration in fhem "bc6100" bzw. "983000". Diese funktionieren hier aber nicht.

dewenni, 2025-02-27:
Ich muss mich mit den Seriennummern nochmal etwas auseinandersetzen und versuchen herauszufinden warum manche Seriennummern scheinbar nicht funktionieren und welches Muster es da gibt.

> Ich nutze aus der fhem Diskussion um die sduino-Integration in fhem "bc6100" bzw. "983000". Diese funktionieren hier aber nicht.

hier gibt es aber auch eine Diskrepanz zu dem Log:
[image]

bc61**00** zu **0**bc61**0** 

Die Seriennummer hat insgesamt 8 Stellen. Die ersten 6 werden vom User vorgegeben und die letzten beiden Stellen werden für den Kanal benutzt.

wunderbaum, 2025-02-27:
> Welche Seriennummer hattest du denn zuvor benutzt, die nicht funktionierte? Irgendwo stand mal, dass die durch 16 teilbar seien muss. Ich weiß aber nicht ob das tatsächlich so sein muss.

Das stand irgendwo und es ist wohl auch was dran, wobei auch ein _ist-durch-16-teilbar_ nicht automatisch zu einer funktionierenden Serial führt. Warum auch immer.

closed 2025-03-02 by dewenni