dewenni/ESP32-Jarolift-Controller #37: [BUG] No movement of blinds after learning from remote

tomarrell, 2025-03-01:
Firstly, I just want to say thank you greatly for building this, it's a really neat solution!

### Is this bug present in the latest release?

- [x] I am using the **latest release**.
- [ ] I am not using the latest release (please update and retest before reporting the bug).

### Hardware Information

- **ESP32 Type**: [ESP32-WROOM-32
](https://www.amazon.de/dp/B0D4QYC4YM?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1)
- [CC1101](https://www.amazon.de/dp/B07YX92NMP?ref=ppx_yo2ov_dt_b_fed_asin_title)

### Bug Description

The setup goes smoothly, up until attempting to learn a remote. I press the up + down buttons on the existing remote, followed by the stop button 8 times. The motor proceeds to jiggle. I then quickly press the "learn" button on the web UI.

However, when I attempt to use the new remote from the channels screen, nothing happens. The existing remote functions, and I notice that when I trigger a few commands, it takes a number of presses of the existing remote before it functions again. 

But the ESP32 remote never functions, I never get any movement from the blinds.

I have also attempted to learn after removing power from the motors. I removed power, restored it, the motors jiggled and I clicked "learn". However same result. No movement of the motors when issuing a command.

### Additional Information

The wiring setup is exactly the same as pictured in the README. I have checked all the GPIO connections are correct.

Logs of the entire interaction. Started with a completely fresh setup.

```
[01.03.2025 - 19:48:20] D WEB: Received - Element ID: p10_log_refresh_btn = true
[01.03.2025 - 19:48:20] D JARO-LIB: (INF3) rx_device_key_lsb: 0x0bd1ce69, rx_device_key_msb: 0xa165361e, decoded: 0x3753e308
[01.03.2025 - 19:48:20] D JARO-LIB: (INF2) RSSI: 76, counter: 8
[01.03.2025 - 19:48:20] D JARO-LIB: CC1101_RSSI: 76
[01.03.2025 - 19:48:20] D JARO-LIB: (INF1) serial: 0x046e7e03, rx_function: 0x8, rx_disc_low: 55, rx_disc_high: 0
[01.03.2025 - 19:48:20] D JARO-LIB: decoded devicekey: 0x3753e308
[01.03.2025 - 19:48:20] D JARO-LIB: received devicekey low: 0x0bd1ce69 // high: 0xa165361e
[01.03.2025 - 19:48:17] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4b376b58, rx_device_key_msb: 0xa523ac0b, decoded: 0xe743debc
[01.03.2025 - 19:48:17] D JARO-LIB: (INF2) RSSI: 75, counter: 188
[01.03.2025 - 19:48:17] D JARO-LIB: CC1101_RSSI: 75
[01.03.2025 - 19:48:17] D JARO-LIB: (INF1) serial: 0x02373f01, rx_function: 0x2, rx_disc_low: 231, rx_disc_high: 0
[01.03.2025 - 19:48:17] D JARO-LIB: decoded devicekey: 0xe743debc
[01.03.2025 - 19:48:17] D JARO-LIB: received devicekey low: 0x4b376b58 // high: 0xa523ac0b
[01.03.2025 - 19:48:16] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4b376b58, rx_device_key_msb: 0xa523ac0b, decoded: 0xea025f45
[01.03.2025 - 19:48:16] D JARO-LIB: (INF2) RSSI: 74, counter: 69
[01.03.2025 - 19:48:16] D JARO-LIB: CC1101_RSSI: 74
[01.03.2025 - 19:48:16] D JARO-LIB: (INF1) serial: 0x02373f01, rx_function: 0x4, rx_disc_low: 234, rx_disc_high: 0
[01.03.2025 - 19:48:16] D JARO-LIB: decoded devicekey: 0xea025f45
[01.03.2025 - 19:48:16] D JARO-LIB: received devicekey low: 0x4b376b58 // high: 0xa523ac0b
[01.03.2025 - 19:48:11] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4b376b58, rx_device_key_msb: 0xa523ac0b, decoded: 0xe78301ba
[01.03.2025 - 19:48:11] D JARO-LIB: (INF2) RSSI: 76, counter: 186
[01.03.2025 - 19:48:11] D JARO-LIB: CC1101_RSSI: 76
[01.03.2025 - 19:48:11] D JARO-LIB: (INF1) serial: 0x02373f01, rx_function: 0x8, rx_disc_low: 231, rx_disc_high: 255
[01.03.2025 - 19:48:11] D JARO-LIB: decoded devicekey: 0xe78301ba
[01.03.2025 - 19:48:11] D JARO-LIB: received devicekey low: 0x4b376b58 // high: 0xa523ac0b
[01.03.2025 - 19:48:05] D WEB: Received - Element ID: p10_log_refresh_btn = true
[01.03.2025 - 19:48:00] I JARO: execute cmd: DOWN - channel: 1
[01.03.2025 - 19:48:00] D JARO-LIB: Device Counter: 4
[01.03.2025 - 19:48:00] D JARO-LIB: created devicekey low: 0xbc54a53a // high: 0x183dba7e
[01.03.2025 - 19:48:00] D JARO-LIB: get serial: 0x00000000 for channel: 1
[01.03.2025 - 19:48:00] D JARO: add single cmd to buffer: 1, 1
[01.03.2025 - 19:48:00] I WEB: cmd DOWN - channel 1
[01.03.2025 - 19:48:00] D WEB: Received - Element ID: p01_down_0 = true
[01.03.2025 - 19:48:00] I JARO: execute cmd: STOP - channel: 1
[01.03.2025 - 19:48:00] D JARO-LIB: Device Counter: 3
[01.03.2025 - 19:47:59] D JARO-LIB: created devicekey low: 0xbc54a53a // high: 0x183dba7e
[01.03.2025 - 19:47:59] D JARO-LIB: get serial: 0x00000000 for channel: 1
[01.03.2025 - 19:47:59] D JARO: add single cmd to buffer: 2, 1
[01.03.2025 - 19:47:59] I WEB: cmd STOP - channel 1
[01.03.2025 - 19:47:59] D WEB: Received - Element ID: p01_stop_0 = true
[01.03.2025 - 19:47:58] I JARO: execute cmd: DOWN - channel: 1
[01.03.2025 - 19:47:58] D JARO-LIB: Device Counter: 2
[01.03.2025 - 19:47:58] D JARO-LIB: created devicekey low: 0xbc54a53a // high: 0x183dba7e
[01.03.2025 - 19:47:58] D JARO-LIB: get serial: 0x00000000 for channel: 1
[01.03.2025 - 19:47:58] D JARO: add single cmd to buffer: 1, 1
[01.03.2025 - 19:47:58] I WEB: cmd DOWN - channel 1
[01.03.2025 - 19:47:58] D WEB: Received - Element ID: p01_down_0 = true
[01.03.2025 - 19:47:58] I JARO: execute cmd: UP - channel: 1
[01.03.2025 - 19:47:58] D JARO-LIB: Device Counter: 1
[01.03.2025 - 19:47:57] D JARO-LIB: created devicekey low: 0xbc54a53a // high: 0x183dba7e
[01.03.2025 - 19:47:57] D JARO-LIB: get serial: 0x00000000 for channel: 1
[01.03.2025 - 19:47:57] D JARO: add single cmd to buffer: 0, 1
[01.03.2025 - 19:47:57] I WEB: cmd UP - channel 1
[01.03.2025 - 19:47:57] D WEB: Received - Element ID: p01_up_0 = true
[01.03.2025 - 19:47:51] I JARO: execute cmd: LEARN - channel: 1
[01.03.2025 - 19:47:51] I JARO-LIB: Channel learned!
[01.03.2025 - 19:47:51] D JARO-LIB: Device Counter: 0
[01.03.2025 - 19:47:49] D JARO-LIB: created devicekey low: 0xbc54a53a // high: 0x183dba7e
[01.03.2025 - 19:47:49] D JARO-LIB: get serial: 0
[01.03.2025 - 19:47:49] D JARO-LIB: Device Counter from EEPROM: 0
[01.03.2025 - 19:47:49] D JARO-LIB: get serial: 0x00000000 for channel: 1
[01.03.2025 - 19:47:49] I JARO-LIB: putting channel 1 into learn mode ...
[01.03.2025 - 19:47:49] I WEB: cmd LEARN - channel 1
[01.03.2025 - 19:47:49] D JARO: add single cmd to buffer: 5, 1
[01.03.2025 - 19:47:49] D WEB: Received - Element ID: p12_learn_0 = true
[01.03.2025 - 19:47:14] D WEB: Received - Element ID: p10_log_refresh_btn = true
[01.03.2025 - 19:47:12] I WEB: web-client disconnected
[01.03.2025 - 19:47:11] D WEB: updateAllElements()
[01.03.2025 - 19:47:11] I WEB: web-client connected - IP:192.168.178.46
[01.03.2025 - 19:47:11] D WEB: contend not changed: /main.js
[01.03.2025 - 19:47:11] D WEB: contend not changed: /main.css
[01.03.2025 - 19:47:11] D ESU-MRD: Cleared reset state
[01.03.2025 - 19:47:11] D ESU-MRD: Timeout reached, resetting count
[01.03.2025 - 19:47:11] D WEB: contend not changed: /
[01.03.2025 - 19:47:09] D WEB: updateAllElements()
[01.03.2025 - 19:47:09] I WEB: web-client connected - IP:192.168.178.46
[01.03.2025 - 19:47:07] I TELNET: Telnet Server running!
[01.03.2025 - 19:47:07] I JARO: read Device Counter from FLASH: 0
[01.03.2025 - 19:47:07] I JARO: CC1101 Transceiver connected!
[01.03.2025 - 19:47:07] I SETUP: IP address: 192.168.178.41
[01.03.2025 - 19:47:07] I SETUP: WiFi connected
[01.03.2025 - 19:47:06] I SETUP: Connected to AP successfully!
[01.03.2025 - 19:47:06] I JARO-LIB: start CC1101 setup
[01.03.2025 - 19:47:06] D JARO-LIB: set base serial: 0x00000000
[01.03.2025 - 19:47:06] I JARO: initializing the CC1101 Transceiver
[01.03.2025 - 18:47:06] I SETUP: WiFi Mode STA - Trying connect to: FRITZ!Box 7590 ZR 2_4Ghz
[01.03.2025 - 18:47:06] I ESU-WDT: Watchdog timer initialized for Loop Task
[01.03.2025 - 18:47:06] I MSG: LogLevel: ESP_LOG_DEBUG
[01.03.2025 - 18:47:06] I CFG: LittleFS successfully started
```

[image]

[image]

[image]

dewenni, 2025-03-02:
Hi,

it seems that you haven't entered the Jarolift MSB and LSB Key.
And also there is not Serialnumber set.

Please Google the information for the Jarolift Keys. Keywords: Jarolift, MSB, LSB, Pastebin

For the Serialnumber you can use for example `000010`

tomarrell, 2025-03-02:
@dewenni Ah brilliant, that's solved it! Thanks a lot. For some reason, I assumed they may be populated during the pairing process.

Nochmals vielen Dank!

closed 2025-03-02 by tomarrell