dewenni/ESP32-Jarolift-Controller #38: [BUG]

gtsy7, 2025-03-02:
can you send me an email and I will send you a photo?

### Is this bug present in the latest release?

- [ ] I am using the **latest release**.
- [ ] I am not using the latest release (please update and retest before reporting the bug).

### Hardware Information

- **ESP32 Type**: [e.g., ESP32-WROOM-32, ...]

### Bug Description

**Describe the problem clearly and concisely:**

- What is the expected behavior?
- What actually happens?
- Steps to reproduce the issue (if applicable):

### Additional Information

- Attach logs, screenshots, or code snippets if relevant.

closed 2025-03-02 by dewenni

gtsy7, 2025-03-02:
Can you send me an email and I'll send you a photo?

gtsy7, 2025-03-02:
Kodnet7@gmail.com

gtsy7, 2025-03-02:
If you write here, will you receive my letter with a photo?

Closed #38 <https://github.com/dewenni/ESP32-Jarolift-Controller/issues/38>

dewenni, 2025-03-02:
What is your problem?

You have not provided any description at all.

You can also post a photo here!

gtsy7, 2025-03-03:
Send foto
[image: dewenni]

gtsy7, 2025-03-04:
it will be possible to do so?

Send foto