dewenni/ESP32-Jarolift-Controller #39: [FEATURE] Unterstützung von mehr als 16 Geräten

blissi, 2025-03-04:
Hi,
zuerst einmal mega, dass du das auf ESP32 portiert und verbessert hast! Ich kenne den originalen Code auch sehr gut und hatte einmal damit angefangen, dann aber mangels Zeit nicht fertig gestellt und bin stattdessen bei einem Sduino gelandet, den ich auf ESP32 portiert habe. Das funktioniert auch sehr gut, aber was nervt ist, dass ich fhem als Mittler nutzen muss zwischen HA und dem Sduino.

Was noch ein "Relikt" von Mad Martins Code ist, ist die eine, feste Serial Nummer und damit die Limitierung auf 16 Geräte. Hast du das schon auf dem Schirm und vor es zu fixen?

LG

blissi, 2025-03-05:
@dewenni  Genau so ist es. Eine Serial ist sozusagen eine Fernbedienung. Mit zwei Serials hat man deshalb auch zwei Sets an Geräten, die Gruppen untereinander bilden können - nicht übergreifend.

Die Herausforderung wird hauptsächlich sein, das irgendwie gescheit im UI hinzubekommen. Ich konnte mir die Web Demo UI nicht anschauen (Menu funktioniert nicht). Wenns mir reicht, flashe ich meinen ESP mit dem Sduino am WE mal und schaue mir das live an.

dewenni, 2025-03-05:
Das die Demo WebUI bei dir nicht funktioniert, wundert mich etwas.
Kannst du das ggf mal mit verschiedenen Browsern testen?
Zur Not auch mal die Entwickler Tools des Browsers öffnen und schauen ob offensichtliche Fehler angezeigt werden.

Bei mir funktioniert es...

blissi, 2025-03-05:
Das kommt bei mir (Edge und Chrome):
[image]

Klick auf Menupunkte macht nichts, ich sehe immer nur die Startseite mit den Channels.

blissi, 2025-03-08:
Ich habe es jetzt probehalber mal auf meinen ESP geladen und alle notwendigen Einstellungen vorgenommen.
Leider kann ich keinen Rolladen anlernen: AUF+AB und dann 8x Stop, der Rolladen vibriert. Dann schnell auf die Lernen-Taste, aber der Rolladen macht nichts. Ich habs mehrere male probiert, leider hat es nicht geklappt. In den Logs sehe ich auch die Tastendrücke meines Wandsenders, inkl. aufsteigendem Counter -> MSB und LSB sind korrekt und CC1101 auch richtig konfiguriert.

Hast du eine Idee, was ich checken könnte?

```
[08.03.2025 - 10:16:19] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:19] D JARO-LIB: decoded devicekey: 0x20050b1f
[08.03.2025 - 10:16:19] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0xa, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:19] D JARO-LIB: CC1101_RSSI: 97
[08.03.2025 - 10:16:19] D JARO-LIB: (INF2) RSSI: 97, counter: 31
[08.03.2025 - 10:16:19] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b1f
[08.03.2025 - 10:16:19] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:19] D JARO-LIB: decoded devicekey: 0x20050b20
[08.03.2025 - 10:16:19] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0x4, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:19] D JARO-LIB: CC1101_RSSI: 94
[08.03.2025 - 10:16:19] D JARO-LIB: (INF2) RSSI: 94, counter: 32
[08.03.2025 - 10:16:19] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b20
[08.03.2025 - 10:16:20] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:20] D JARO-LIB: decoded devicekey: 0x20050b21
[08.03.2025 - 10:16:20] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0x4, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:20] D JARO-LIB: CC1101_RSSI: 94
[08.03.2025 - 10:16:20] D JARO-LIB: (INF2) RSSI: 94, counter: 33
[08.03.2025 - 10:16:20] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b21
[08.03.2025 - 10:16:20] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:20] D JARO-LIB: decoded devicekey: 0x20050b22
[08.03.2025 - 10:16:20] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0x4, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:20] D JARO-LIB: CC1101_RSSI: 94
[08.03.2025 - 10:16:20] D JARO-LIB: (INF2) RSSI: 94, counter: 34
[08.03.2025 - 10:16:20] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b22
[08.03.2025 - 10:16:21] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:21] D JARO-LIB: decoded devicekey: 0x20050b24
[08.03.2025 - 10:16:21] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0x4, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:21] D JARO-LIB: CC1101_RSSI: 95
[08.03.2025 - 10:16:21] D JARO-LIB: (INF2) RSSI: 95, counter: 36
[08.03.2025 - 10:16:21] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b24
[08.03.2025 - 10:16:22] D JARO-LIB: received devicekey low: 0x4e61c3d1 // high: 0x960934cd
[08.03.2025 - 10:16:22] D JARO-LIB: decoded devicekey: 0x20050b25
[08.03.2025 - 10:16:22] D JARO-LIB: (INF1) serial: 0x058d3905, rx_function: 0x4, rx_disc_low: 32, rx_disc_high: 0
[08.03.2025 - 10:16:22] D JARO-LIB: CC1101_RSSI: 93
[08.03.2025 - 10:16:22] D JARO-LIB: (INF2) RSSI: 93, counter: 37
[08.03.2025 - 10:16:22] D JARO-LIB: (INF3) rx_device_key_lsb: 0x4e61c3d1, rx_device_key_msb: 0x960934cd, decoded: 0x20050b25
[08.03.2025 - 10:16:25] D WEB: Received - Element ID: p12_learn_1 = true
[08.03.2025 - 10:16:25] D JARO: add single cmd to buffer: 5, 2
[08.03.2025 - 10:16:25] I WEB: cmd LEARN - channel 2
[08.03.2025 - 10:16:25] I JARO-LIB: putting channel 2 into learn mode ...
[08.03.2025 - 10:16:25] D JARO-LIB: get serial: 0xa0001001 for channel: 2
[08.03.2025 - 10:16:25] D JARO-LIB: Device Counter from EEPROM: 0
[08.03.2025 - 10:16:25] D JARO-LIB: get serial: 0xa0001001
[08.03.2025 - 10:16:25] D JARO-LIB: created devicekey low: 0x59ae1f0d // high: 0x95ab9bc7
[08.03.2025 - 10:16:27] D JARO-LIB: Device Counter: 0
[08.03.2025 - 10:16:27] I JARO-LIB: Channel learned!
[08.03.2025 - 10:16:27] I JARO: execute cmd: LEARN - channel: 2
[08.03.2025 - 10:16:27] D WEB: Received - Element ID: p12_learn_1 = true
[08.03.2025 - 10:16:27] D JARO: add single cmd to buffer: 5, 2
[08.03.2025 - 10:16:27] I WEB: cmd LEARN - channel 2
[08.03.2025 - 10:16:27] I JARO-LIB: putting channel 2 into learn mode ...
[08.03.2025 - 10:16:27] D JARO-LIB: get serial: 0xa0001001 for channel: 2
[08.03.2025 - 10:16:27] D JARO-LIB: Device Counter from EEPROM: 0
[08.03.2025 - 10:16:27] D JARO-LIB: get serial: 0xa0001001
[08.03.2025 - 10:16:27] D JARO-LIB: created devicekey low: 0x59ae1f0d // high: 0x95ab9bc7
[08.03.2025 - 10:16:28] D JARO-LIB: Device Counter: 0
[08.03.2025 - 10:16:28] I JARO-LIB: Channel learned!
[08.03.2025 - 10:16:28] I JARO: execute cmd: LEARN - channel: 2
[08.03.2025 - 10:16:28] D WEB: Received - Element ID: p12_learn_1 = true
[08.03.2025 - 10:16:28] D JARO: add single cmd to buffer: 5, 2
[08.03.2025 - 10:16:28] I WEB: cmd LEARN - channel 2
[08.03.2025 - 10:16:28] I JARO-LIB: putting channel 2 into learn mode ...
[08.03.2025 - 10:16:28] D JARO-LIB: get serial: 0xa0001001 for channel: 2
[08.03.2025 - 10:16:28] D JARO-LIB: Device Counter from EEPROM: 0
[08.03.2025 - 10:16:28] D JARO-LIB: get serial: 0xa0001001
[08.03.2025 - 10:16:28] D JARO-LIB: created devicekey low: 0x59ae1f0d // high: 0x95ab9bc7
[08.03.2025 - 10:16:30] D JARO-LIB: Device Counter: 0
[08.03.2025 - 10:16:30] I JARO-LIB: Channel learned!
[08.03.2025 - 10:16:30] I JARO: execute cmd: LEARN - channel: 2
[08.03.2025 - 10:16:32] D WEB: Received - Element ID: p10_log_refresh_btn = true
```

dewenni, 2025-03-08:
@blissi du hast als Seriennummer jetzt `a00010` verwendet, oder?
Teste bitte mal `000010`

blissi, 2025-03-08:
Damit hatte ich es davor schon probiert

dewenni, 2025-03-08:
tja komisch.
Ich habe es gerade mit meinem Test-Setup nochmal probiert.
Seriennummer: 0000ff

- vorhandene Fernbedienung: AUF+AB - 8x Stopp
- Motor vibriert
- Lern Taste im WebUI gedrückt
- Motor vibriert erneut
- funktioniert

Prüfe nochmal MSB und LSB
Muss in dieser Form sein: `27***a9b` und `117***35` (ohne Sternchen natürlich)

dewenni, 2025-03-08:
und noch ein Hinweis. Mir war aufgefallen, dass wenn man MSB, LSB oder Serial eingibt und danach direkt auf den "Lock" button drückt. Dann wird die Eingabe nicht immer übernommen. Das habe ich mit der nächsten Version verbessert.

blissi, 2025-03-08:
So habs nochmal ausprobiert und jetzt hats geklappt - danke für den Tipp, vielleicht lag es ja tatsächlich daran.

dewenni, 2025-03-08:
ok, danke für die Rückmeldung.
Denkst du es lag daran, dass der Device Counter 0 war oder daran das MSB, LSB nicht korrekt gespeichert waren?

blissi, 2025-03-08:
ich bin mir nicht sicher. ich hatte davor noch mein case geöffnet und die Kabel zum CC1101 gezogen und neu eingesteckt. Vielleicht lags auch daran