dewenni/ESP32-Jarolift-Controller #40: [BUG] Firmware Update schlägt fehl

markus-hi, 2025-03-09:
### Wurde der Fehler in der neuesten Version getestet?

- [X ] Ich nutze die Version 1.63 und mächte auf 1.7.0 updaten.

Bisherige OTA Updates liefen immer problemlos durch.

### Hardware-Informationen

- ESP32 Mini

### Fehlerbeschreibung

Update auf 1.7.0 schlägt fehl, letzte Zeilen aus Logdatei:

[09.03.2025 - 18:53:21] I GithubReleaseOTA: Written 1598636/1599680 bytes
[09.03.2025 - 18:53:21] I GithubReleaseOTA: Written 1599660/1599680 bytes
[09.03.2025 - 18:53:21] I GithubReleaseOTA: Written 1599680/1599680 bytes
[09.03.2025 - 18:53:21] E boot_comm: mismatch chip ID, expected 0, found 5
[09.03.2025 - 18:53:21] E GithubReleaseOTA: Failed to end OTA update
[09.03.2025 - 18:53:21] E GITHUB: Firmware update failed: 5
[09.03.2025 - 18:53:21] E WEB: GitHub OTA-Update failed: End error
[09.03.2025 - 18:53:21] I ESU-WDT: Watchdog timer initialized for Loop Task
[09.03.2025 - 18:53:24] D WEB: Received - Element ID: p00_ota_failed_btn = true

DirkHer, 2025-03-09:
Bei mir das gleiche, auch von 1.6.3 kommend. Gleiche Hardware.

dewenni, 2025-03-09:
Danke für die Rückmeldung. Problem erkannt!
Durch die Unterstützung von mehreren ESP32 Varianten (S2, S3, C3) habe ich jetzt auch mehrere Update Dateien zur Verfügung gestellt. 
Die Software überprüft auch jetzt welcher ESP-Chip verwendet wird und wählt anhand von dem "Chip-Namen" auch die Update Datei.

Derzeit werden folgende Chips unterstützt:

- ESP32
- ESP32s2
- ESP32s3
- ESP32c3

Wenn der Chip-Name nun aber einfach nur "ESP32" ist, also dann wenn es ein "normaler" ESP32 ist, dann kommt dieser Name aber leider auch in allen andern Dateinamen vor. In der Folge wird dann die falsche Update Datei verwendet und es kommt zu dem Fehler. Keine Ahnung warum mir das nicht gleich aufgefallen ist. Ich muss die Abfrage hier anpassen.

Als Übergangslösung habe ich nun einfach im Release 1.7.0 die anderen OTA-Update Dateien einfach mal wieder gelöscht.
Jetzt gibt es nur noch die  `esp32s3_jarolift_ota_v1.7.0.bin`
Damit sollte das Update für die allermeisten funktionieren.
Mit dem nächsten Update passe ich die Abfrage an. Dann sollte das auch mit den anderen funktionieren.

closed 2025-03-23 by dewenni