dewenni/ESP32-Jarolift-Controller #47: [BUG]

Oxi75, 2025-03-23:
Zunächst mal ein riesen Lob für dieses geniale Projekt! Ich habe schon seit langem auf so etwas gewartet und meine Entscheidung verflucht die TDEF Motoren bei mir verbaut zu haben. Nun kann ich sie doch noch über mein Handy steuern!! :-) 

### Wurde der Fehler in der neuesten Version getestet?

- [x] Ich nutze die **neueste Version** (1.9)

### Hardware-Informationen

- **ESP32-Typ**: MH-ET DEVKIT 32

### Fehlerbeschreibung

Ich bin mir nicht sicher, ob das Fehlverhalten wirklich ein Bug oder ein anderes Problem (Konfiguration, ..) ist.

Ich habe eine Reihe von Rolläden eingelernt. Der Rolladen auf Kanal 10 und testweise auch auf Kanal 16 fährt nach unten, obwohl ich nach oben drücke und umgekehrt. Die Steuerung über die original-Fernbedienung funktioniert allerdings wie gewünscht. Alle anderen Rolläden fahren, soweit ich das bisher sehen konnte korrekt.

Besteht die Möglichkeit, dass die Button-Funktionen vertauscht sind? Meines Wissen hat die Drehrichtung des Motors nichts mit der Fernbedienung zu tun, sondern wird im Motor selber gespeichert. Alle eingelernten Fernbedienungen müssten sich somit identisch verhalten.

dewenni, 2025-03-23:
Das kann ich spontan nicht nachvollziehen.

Kannst du das bitte mal über verschiedene Wege prüfen und Rückmeldung geben
- Ansteuerung über die Seite "Kanäle" im WebUI
- Ansteuerung über die Seite "Service" im WebUI
- Ansteuerung über MQTT

Oxi75, 2025-03-23:
Hi! Ich muss noch mal nachschauen - ich befürchte, dass doch irgendwas an meinem System nicht stimmt - warum aber die beiden Fernbedienungen anders reagieren verstehe ich noch nicht ganz.
Ich gebe Dir Bescheid.

Eine ganz andere Sache: Das Compilieren für den ESP32c3 schlägt bei mir fehlt. Kannst Du das reproduzieren? ESP32, -S2 und -S3 funktionieren.

dewenni, 2025-03-23:
====================== [SUCCESS] Took 19.69 seconds ======================

Environment    Status    Duration
-------------  --------  ------------
esp32c3        SUCCESS   00:00:19.689
====================== 1 succeeded in 00:00:19.689 ======================

Oxi75, 2025-03-23:
Hallo Sven,

Du kannst den Bugreport schließen. Das war ein typtische 60cm Fehler (also der User vorm PC).
Warum der C3 bei mir nicht will muss ich mir noch mal genauer anschauen.

Viele Grüße und noch einen schönen Sonntag!!

closed 2025-03-23 by Oxi75

dewenni, 2025-03-23:
Welchen Fehler bekommst du denn?

Oxi75, 2025-03-24:
Environment    Status    Duration
-------------  --------  ------------
esp32          SUCCESS   00:01:30.163
esp32s2        SUCCESS   00:01:17.255
esp32s3        SUCCESS   00:01:22.401
esp32c3        FAILED    00:00:18.144

riscv32-esp-elf-g++: fatal error: cannot execute 'C:/Users/Andreas/.platformio/packages/toolchain-riscv32-esp/bin/../lib/gcc/riscv32-esp-elf/13.2.0/../../../../riscv32-esp-elf/bin/as.exe': CreateProcess: No such file or directory

Das scheint mir auch eher ein Problem mit der Environment zu sein aber verstehen tue ich ihn nicht (der Pfad existiert und dort auch as.exe). Habe schon so ziemlich alles mögliche versucht einschließlich Löschen und Neuinstallation von VSCode inkl. PlatformIO.

Das alles würde mich ja nicht so wirklich stören, hätte ich hier nicht noch einen arbeitslosen c3-Mini liegen, dem ich gerne meine Rolladensteuerung überlassen würde.

wunderbaum, 2025-03-24:
Unter Windows werden die Pfade zu lang und dann klappt das nicht.
Abhilfe schafft ein:
```
[platformio]
core_dir = c:.platformio
workspace_dir = c:.pio
```
ganz am Anfang der platformio.ini.

Oxi75, 2025-03-24:
Danke Dir! Das mit den Pfaden habe ich irgendwann auch gelesen. Lustigerweise funktioniert es inzwischen auch am angestammten Platz. Aber der Fehler kommt bestimmt irgendwann wieder, dann werde ich mich hoffentlich daran erinnern ;-)