dewenni/ESP32-Jarolift-Controller #48: [BUG] OTA Updatefehler

wwwilly63, 2025-03-24:
### Wurde der Fehler in der neuesten Version getestet?

- [ ] Ich nutze die **neueste Version**.
- [ X] Ich nutze die 1.6.3 Version und möchte auf 1.9 updaten
### Hardware-Informationen

- ESP32 mini

### Fehlerbeschreibung

**Update schlägt fehl**

[24.03.2025 - 12:35:36] I GithubReleaseOTA: Written 1605546/1606384 bytes
[24.03.2025 - 12:35:36] I GithubReleaseOTA: Written 1606384/1606384 bytes
[24.03.2025 - 12:35:36] E boot_comm: mismatch chip ID, expected 0, found 5
[24.03.2025 - 12:35:36] E GithubReleaseOTA: Failed to end OTA update
[24.03.2025 - 12:35:36] E GITHUB: Firmware update failed: 5
[24.03.2025 - 12:35:36] E WEB: GitHub OTA-Update failed: End error
[24.03.2025 - 12:35:36] I ESU-WDT: Watchdog timer initialized for Loop Task
[24.03.2025 - 12:35:48] D WEB: Received - Element ID: p00_ota_failed_btn = true 

### Zusätzliche Informationen

wurde bereits gemeldet Firmware Update schlägt fehl #40, Version 1.6.3 auf 1.7

bart83-tech, 2025-03-30:
Wurde der Fehler in der neuesten Version getestet?

        Ich nutze die neueste Version.
        [ X] Ich nutze die 1.6.3 Version und möchte auf 1.9 updaten

Hardware-Informationen

    ESP32 D1 mini 

Fehlerbeschreibung

Update schlägt fehl

Hallo, ich habe dasselbe Problem. Bis zum Update auf Version 1.6.3 ging alles problemlos. 
Ich nutze einen ESP32 D1 Mini.

Hier mein log. Ich wäre auch über Ideen dankbar. 

[30.03.2025 - 14:12:29] E WEB: GitHub OTA-Update failed: End error
[30.03.2025 - 14:12:29] E GITHUB: Firmware update failed: 5
[30.03.2025 - 14:12:29] E GithubReleaseOTA: Failed to end OTA update
[30.03.2025 - 14:12:29] E boot_comm: mismatch chip ID, expected 0, found 5

bart83-tech, 2025-04-01:
Ich habe mit dem OTA Update, die Version auf 1.7.0 (https://github.com/dewenni/ESP32-Jarolift-Controller/releases/tag/v1.7.0) Updaten können und dann auf. 1.9.0. In dieser Reihenfolge hat es funktioniert!

closed 2025-04-01 by dewenni