dewenni/ESP32-Jarolift-Controller #50: [BUG] w5500 verify chip ID failed

atze09, 2025-04-09:
### Wurde der Fehler in der neuesten Version getestet?

- [ X] Ich nutze die **neueste Version**.
- [ ] Ich nutze nicht die neueste Version (bitte zuerst aktualisieren und erneut testen, bevor der Fehler gemeldet wird).

### Hardware-Informationen

- ESP32-D0WD-V3 : [z. B. ESP32-WROOM-32, ...]

### Fehlerbeschreibung

**Beschreibe das Problem klar und präzise:**

- Was ist das erwartete Verhalten?
- ETH MOdul wird nicht erkannt, habe 3 verschiedene Module getestet
- Was passiert tatsächlich?
- Schritte zur Reproduktion des Fehlers (falls zutreffend):
- Verkabeln und in der GUI zuweisen.

### Zusätzliche Informationen

[09.04.2025 - 06:54:51] E esp_eth: esp_eth_driver_install(228): init mac failed
[09.04.2025 - 06:54:51] E w5500.mac: emac_w5500_init(826): verify chip ID failed
[09.04.2025 - 06:54:51] E w5500.mac: W5500 version mismatched, expected 0x04, got 0x00

Vielen Dank schon einmal.

dewenni, 2025-04-09:
Welche Module hast du denn genau verwendet?

Hier gibt es funktionierende Referenzen:
https://github.com/dewenni/ESP32-Jarolift-Controller/discussions/34

atze09, 2025-04-09:
Hab es nochmal abgeglichen, danke es funktioniert jetzt.
Super Arbeit, danke dafür!

dewenni, 2025-04-09:
woran lag es jetzt?
Verkabelung oder GPIO Konfiguration?

atze09, 2025-04-09:
Habe beides auf das template angepasst denke die konfig.