dewenni/ESP32-Jarolift-Controller #51: [BUG] no signal from existing jarolift remote

bahlsino, 2025-04-11:
### Is this bug present in the latest release?

- [x] I am using the **latest release**.
          esp32_jarolift_flash_v1.9.0.bin (from the release section)
- [ ] I am not using the latest release (please update and retest before reporting the bug).

### Hardware Information
       - Esp32 d1 mini
       - CC1101 Funkmodul Modul RUIZHI CC1101
       - 8 channel remote TDRC08W

### Bug Description

**Describe the problem clearly and concisely:**
   Send works as expected for all of my 8 shutters (even if remote is far away from controller)
   But nothing received from original remote (in WebUI-Logger or MQTT)
   Tested with remote near to controller and logger Mode set to DEBUG

- What is the expected behavior?
    The serial number can be found in the logbook by pressing the remote control near the controller.

- What actually happens?
    no message in the log or via mqtt
 
- Steps to reproduce the issue (if applicable):

### Additional Information

- Attach logs, screenshots, or code snippets if relevant.

PersistentCloud, 2025-05-20:
I've tried imto downgrade the firmware to 1.7.0 but still no success. My remote is not shown in the logs after using it multiple times

`[20.05.2025 - 09:42:47] I CFG: LittleFS successfully started
[20.05.2025 - 09:42:48] I CFG: config successfully saved to file: /config.json - Version: 1
[20.05.2025 - 09:42:48] I CFG: config file was updated from version 2 to version: 1
[20.05.2025 - 09:42:48] I MSG: LogLevel: ESP_LOG_DEBUG
[20.05.2025 - 09:42:48] I ESU-WDT: Watchdog timer initialized for Loop Task
[20.05.2025 - 09:42:48] I SETUP: WiFi Mode STA - Trying connect to: DesIsUnserWLAN
[20.05.2025 - 11:42:48] I SETUP: ESP-ChipSeries: ESP32
[20.05.2025 - 11:42:48] I SETUP: ESP-ChipModel: ESP32-D0WD-V3
[20.05.2025 - 11:42:48] I SETUP: ESP-ChipRevision: 301
[20.05.2025 - 11:42:48] I SETUP: ESP-CpuFreq: 240
[20.05.2025 - 11:42:48] I SETUP: ESP-FlashChipSize: 4.00 MB
[20.05.2025 - 11:42:48] I JARO: initializing the CC1101 Transceiver
[20.05.2025 - 11:42:48] D JARO-LIB: set base serial: 0x00000010
[20.05.2025 - 11:42:48] I JARO-LIB: start CC1101 setup
[20.05.2025 - 11:42:49] I JARO: CC1101 Transceiver connected!
[20.05.2025 - 11:42:49] I JARO: read Device Counter from FLASH: 29
[20.05.2025 - 11:42:49] I TELNET: Telnet Server running!
[20.05.2025 - 11:42:49] I SETUP: Connected to AP successfully!
[20.05.2025 - 11:42:50] I SETUP: WiFi connected
[20.05.2025 - 11:42:50] I SETUP: IP address: 192.168.1.253
[20.05.2025 - 11:42:50] I MQTT: MQTT setup done!
[20.05.2025 - 11:42:50] I MQTT: MQTT - connection attempt: 1/5
[20.05.2025 - 11:42:51] I MQTT: MQTT connected
[20.05.2025 - 11:42:51] I MQTT: ESP restarted (ota update)
[20.05.2025 - 11:42:52] D ESU-MRD: Timeout reached, resetting count
[20.05.2025 - 11:42:52] D ESU-MRD: Cleared reset state
[20.05.2025 - 11:43:08] D WEBUI: authentication disabled
[20.05.2025 - 11:43:08] D WEBUI: sending: /
[20.05.2025 - 11:43:08] I WEBUI: web-client connected - IP:192.168.1.139
[20.05.2025 - 11:43:08] D WEB: updateAllElements()
[20.05.2025 - 11:43:08] D WEBUI: authentication disabled
[20.05.2025 - 11:43:08] D WEBUI: contend not changed: /lib.css
[20.05.2025 - 11:43:08] D WEBUI: authentication disabled
[20.05.2025 - 11:43:08] D WEBUI: sending: /user.css
[20.05.2025 - 11:43:08] D WEBUI: authentication disabled
[20.05.2025 - 11:43:08] D WEBUI: sending: /lib.js
[20.05.2025 - 11:43:08] D WEBUI: authentication disabled
[20.05.2025 - 11:43:08] D WEBUI: sending: /user.js
[20.05.2025 - 11:43:09] I WEBUI: web-client disconnected
[20.05.2025 - 11:43:10] I WEBUI: web-client connected - IP:192.168.1.139
[20.05.2025 - 11:43:10] D WEB: updateAllElements()
[20.05.2025 - 11:43:21] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:29] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:33] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:33] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:33] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:33] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:53] D WEB: Received - Element ID: p10_log_refresh_btn = true
[20.05.2025 - 11:43:54] D WEB: Received - Element ID: p10_log_refresh_btn = true `

hmsgde, 2025-06-15:
I can confirm the same with remote TDRC16. Sending commands from the controller and the remote control to the shutter works perfect, but no command from the remote control is being showed in the web log (tested with all commands up/down/stop on all channels 1-16).

- ESP32-D0WD-V3 (ESP-WROOM-32), JOY-IT "ESP32 Node MCU Modul", EAN 4250236816104, ESP-WROOM-32)
- CC1101 433Mhz (E07-M1101D, https://www.ebay.de/itm/284904570293 Variante HW-862)
- Jarolift TDRC16
- Firmware esp32_jarolift_flash_v1.9.0.bin

Greets Heiko

PersistentCloud, 2025-06-21:
Hi @hmsgde,

[image]

I have been using MQTT right from the start. My system tab also confirms the active connection to my MQTT server. I am using it together with HomeAssistant. Do you have any more information on how I can try to get it working?

Best regards

PersistentCloud, 2025-06-29:
Hi @hmsgde,

Yes, it's working fine. I've configured it from the beginning as described in your linked post and I'm able to control it from Home Assistant over MQTT. Only my remote is not shown in the logs.

Best regards

PersistentCloud, 2025-08-25:
Updated to v.1.9.0, still not working

dewenni, 2025-10-03:
Unfortunately, receiving signals from original remote controls only works to a very limited extent. For some users, it only works when the remote control is very close, and for others it does not work at all.

Unfortunately, I do not yet know why this is the case.

Sending works very reliably, even over long distances.
Therefore, I would describe receiving as experimental.

wunderbaum, 2025-10-03:
I think, the common cc1101-sets are more or less good at sending, but are often deaf for receiving.
So, don't spend to much energy in trying to detect original remote controls. Replace them by home assistant (or similar) and zigbee-devices (buttons).

EmDee, 2025-10-04:
Makes sense. I like the idea of replacing the original remote with ZigBee controls 💡

dewenni, 2026-11-14:
major improvement with [release v2.0.0](https://github.com/dewenni/ESP32-Jarolift-Controller/releases/tag/v2.0.0)

closed 2026-11-14 by dewenni